Duet3D/RepRapFirmware #994: [Bug]: Feedrate lower than F1.6 causes no motion and RRF stuck at 'Busy'

droftarts, 2024-05-07:
### Duet Forum Discussion Thread

N/A

### Which Duet products are you using?

- [X] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [X] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

RRF 3.5.1

### Duet Web Control Version

DWC 3.5.1

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
```

### Please upload the content of your config.g file.

```gcode
Config.g
```

### What happened?

Setting minimum feed rate with M203 to `M203 I0` results in minimum feed rate set to 0.6mm/min
Sending movement command eg `G1 X1 F1.5` results in no movement and firmware stuck at 'Busy'.
Requested speed is also shown as 0 in Object Model and DWC.

oliof, 2024-05-07:
try setting a non zero value like 0.001 as a workaround, does that help?

-------- Ursprüngliche Nachricht --------

droftarts, 2024-05-07:
@oliof No. I tested with M203 I1 too. Any value for I below 0.6 and M203 is set to I0.6.

dc42, 2024-05-18:
This issue had the same cause as #989. Fixed in 3.5-dev.

closed 2024-05-18 by dc42