Duet3D/RepRapFirmware #995: Large object models cause request issues when M409 and rr_model used at the same time

T3P3, 2024-05-08:
503 errors occur in DWC running duet2 ethernet with a large Object Model, when a PanelDue is also connected.

The proposed solution it to ensure only one of the channels can query the OM at one time.

dc42, 2024-05-25:
Added a mutex so that PanelDue and HTTP don't generate an object model response concurrently. Still need to delay generating that response if insufficient buffer space available.

dc42, 2024-05-28:
Fixed in 3.5-dev.

closed 2024-05-28 by dc42

AlexKid007, 2024-08-22:
Hi @dc42,

today I updated my Duet2 Ethernet to Version 3.5.2. I still get disconnect and reconnect messages very frequently on web interface console page.
 
[image]