Duet3D/RepRapFirmware #997: RRF no longer respects M665 B parameter on linear delta printers

droftarts, 2024-05-10:
As reported here https://forum.duet3d.com/topic/35635/delta-head-going-beyond-bed-limits-3-5-1/, RRF seems to no longer (and may not have done for a while) respect the M665 B parameter, the 'safe printing radius' on delta printers. 

I've just tested this on my delta (Maestro RRF 3.5.0-rc.3+5, can test on 3.5.1 and further back if you want) and it goes outside the bounds set by that parameter. Test at X0 Y0 Z10, with M665 B25, went happily off to Y-60!
```
(Moved to Y-60, reported in DWC and OM)
10/05/2024, 11:49:00 	G1 Y-60
10/05/2024, 11:48:45 	G1 X0 Y0 Z10 F6000
(Axes homed here)
10/05/2024, 11:48:20 	M665
Diagonals 161.671:161.671:161.671, delta radius 81.964, homed height 173.144, bed radius 25.0, X 0.027°, Y 0.355°, Z 0.000°
10/05/2024, 11:48:16 	M665 B25
10/05/2024, 11:47:21 	M665
Diagonals 161.671:161.671:161.671, delta radius 81.964, homed height 173.144, bed radius 75.0, X 0.027°, Y 0.355°, Z 0.000°
```

dc42, 2024-05-17:
The reason is that at GCodes.cpp(2346) the axes passed in parameter axesToLimit are now the axes that have been mentioned explicitly; when previously they were all homed axes. This change was made so that axes owned by other motion systems are not considered. However, LinearDeltaKinematics::LimitPosition only applies limits if all of XYZ are included in that parameter.

dc42, 2024-05-17:
Fix implemented for all affected kinematics. Tested on linear delta kinematics only.

closed 2024-05-17 by dc42