Duet3D/RepRapFirmware #1000: [FeatureRequest] Add M code to configure CAN_0 and CAN_1 on 6HC/6XD

droftarts, 2024-05-13:
Currently, only CAN_1 is supported, configured as a CAN_FD bus. Add options to:
1. Enable CAN_1 as CAN-FD bus, disable CAN_0 - This is the current default
2. Enable CAN_1 as CAN-FD bus, enable CAN_0 as CAN bus - possible in firmware, used by Hangprinter
3. Enable CAN_0 as CAN-FD bus, disable CAN_1 - (dc42 said) There is a 6HC build configuration to use CAN0 instead of CAN1. It should still build, but hasn't been tested in a long time.
4. Enable CAN_0 as CAN-FD bus, enable CAN_1 as CAN bus - (dc42 said) Don't know whether plain CAN is supported on CAN1, it depends on how support was added for normal CAN on the second channel. So may be possible.

(dc42 said) CAN_0 and CAN_1 cannot be enabled as CAN-FD ports at the same time, because the firmware assumes all expansion boards are on the same bus.

T3P3, 2026-09-14:
Forum threads:
https://forum.duet3d.com/topic/38615
https://forum.duet3d.com/topic/36282
https://forum.duet3d.com/topic/34354
https://forum.duet3d.com/topic/31206

chrishamm, 2026-11-02:
Implemented in 2d8d7fd39 for 3.7.1, but differently from the options above. CAN0 is effectively off by default, it only listens and warns about boards connected to it. `M953 P0 A64` turns it into a second CAN-FD bus for the boards with addresses 64 to 125, while all other boards stay on CAN1. Both buses run at the same time.

Regarding the original points:
1. `M953 P0 C1` makes CAN0 a classic CAN bus for the ODrives again. Hangprinter configs need that line from 3.7.1 on.
2. This is the default now.
3. and 4. Swapping the buses is not supported. I removed the old Duet3_CAN0 build config because it only changed the version suffix.

The M953 docs and the CAN connection page have been updated.