Duet3D/RepRapFirmware #1015: Input shaping on linear delta causes Z motor clunking at specific bed locations

droftarts, 2024-06-14:
See https://forum.duet3d.com/topic/35862/delta-z-motor-goes-clunk-regardless-of-motor-or-driver

Z motor (on Z tower) of linear delta 'clunks' when commanded to move between specific points. Reproduced in bench setup using users config.g. Happens in standalone and SBC with RRF 3.5.1, and reportedly with 3.4.2. Disabling IS and rerunning test gcode results in no clunks.

User reports different IS types, and frequencies, still results in clunks.

Tested and reproduced by @droftarts on Duet 3 6HC with RRF 3.5.1, bench test setup. An earlier test with Duet 2 Maestro on RRF 3.5.1 did not elicit any clunks, but movement was limited as testing on real machine.

T3P3, 2025-01-18:
The users reporting the issue have been asked to retest with 3.6beta3

droftarts, 2025-02-17:
I've gone back and tested this 'Delta clunking' issue. I set up a 6HC with 3.5.3 and reproduced the clunking, updated to 3.6.0-beta.4 and reran, no clunking. I don't think we're planning on fixing this in 3.5.x, as IS has been rewritten in 3.6.

If you're facing this issue in 3.5.x, it can be resolved by turning off input shaping.

dc42, 2025-02-20:
As this is fixed in 3.6beta releases and finding/fixing it in 3.5.x is likely to be very time consuming, we're not planning to fix it in 3.5.x.