Duet3D/RepRapFirmware #1016: [FeatureRequest]: Forward Run / Reverse Run support for Spindles in `M950`

benagricola, 2024-06-18:
### Is your feature request related to a problem? Please describe.

Currently, `M950 R0` with 3 pins defined treats the first as PWM speed output, the second as 'spindle on/off' and the third as 'spindle direction'. That means when the spindle is on, the second pin is always active and the state of pin 3 changes based on the direction.

This works for VFD's where there is a Spindle ON input and a Spindle DIR input, but there are quite a lot of VFD's (mine included, Innovance MD200-NC, Delta VFD's follow this pattern too) where this isn't the case - they use separate inputs for Forward RUN / Reverse RUN.

If you configure a Forward RUN / Reverse RUN VFD and connect it to RRF's Enable / Direction pins then the spindle will only start in forward run mode, and when you turn OFF the spindle after switching to reverse mode then the spindle may start in reverse, depending on how the VFD is setup (with 0 RPM input they may run at a low default frequency).
If both forward and reverse run are active at the same time then the spindle will not move. 

### Describe the solution you propose.

The code change to enable the existing pins as Forward / Reverse RUN rather than ENA / DIR is simple (without configuration):

```
--- a/src/Tools/Spindle.cpp
+++ b/src/Tools/Spindle.cpp
@@ -129,6 +129,7 @@ void Spindle::SetRpm(uint32_t rpm) noexcept
        if (state == SpindleState::stopped || rpm == 0)
        {
                onOffPort.WriteDigital(false);
+               reverseNotForwardPort.WriteDigital(false);
                pwmPort.WriteAnalog(idlePwm);
                currentRpm = 0;                                         // current rpm is flagged live, so no need to change seqs.spindles
        }
@@ -143,9 +144,9 @@ void Spindle::SetRpm(uint32_t rpm) noexcept
        else if (state == SpindleState::reverse)
        {
                rpm = constrain<int>(-rpm, -maxRpm, -minRpm);
-               reverseNotForwardPort.WriteDigital(true);
+               onOffPort.WriteDigital(false);
                pwmPort.WriteAnalog(((float)(-rpm - minRpm) / (float)(maxRpm - minRpm)) * (maxPwm - minPwm) + minPwm);
-               onOffPort.WriteDigital(true);
+               reverseNotForwardPort.WriteDigital(true);
                currentRpm = -rpm;                                      // current rpm is flagged live, so no need to change seqs.spindles
        }
 }
```

But it would be great if this could be integrated into the `M950` command as an option, allowing support for VFD's that take this style of inputs.

### Provide any additional context or information.

[Discussion on duet forum](https://forum.duet3d.com/topic/35936/direction-control-in-rrf/6?_=1718621785509)

benagricola, 2024-06-19:
I note there is an `alternateConfig` pin specifier that seems to be mostly undocumented - `*` - I wonder if this could be used to change the behaviour of the spindle enable pin so when alternate config is specified, the enable / direction pins are treated as forward / reverse run.

dc42, 2024-11-04:
PR to 3.6 accepted and merged.

closed 2024-11-04 by dc42