Duet3D/RepRapFirmware #1018: [Bug]: RRF 3.5.2 - LED strips connected to io ports on Sammy C21 (1LC, 1XD) set LEDs incorrectly, all white

droftarts, 2024-06-25:
Forum thread: [https://forum.duet3d.com/topic/35855/led-voron-ws2812e/8](https://forum.duet3d.com/topic/35855/led-voron-ws2812e/8)

I wired up my usual 'test' RGB LEDs (that work well with 3.3V signalling) to a Mini 5+ with 1LC running RRF 3.5.2, using 1LC IO_0.out (the only PWM IO pin on a 1LC). To configure, I sent:
```
M950 E0 C"121.io0.out" U3 T1
```
To turn on just reds, I sent
```
M150 E0 R255 U0 B0 S3
```
However, all three LEDs come on white. After this, any command I send doesn't change this. I think this is a similar timing issue bug as we recently had on the 6HC . 

6HC similar issue: https://github.com/Duet3D/RepRapFirmware/issues/996
Duet 2 similar issue: https://github.com/Duet3D/RepRapFirmware/issues/969

dc42, 2024-08-13:
Fixed in 3.6-dev

dc42, 2024-08-13:
Back-ported to 3.5-dev.

closed 2024-08-13 by dc42