Duet3D/RepRapFirmware #1019: Main board used as expansion board doesn't handle idle current percent

dc42, 2024-07-01:
If a main board is used as an expansion board then changing the idle current percentage on the master board using M906 with the I parameter doesn't affect the idle current percent used on the slave board, which remains the default 30%.

dc42, 2024-07-01:
Provisional fix implemented in 3.6 source code. This could be back-ported to 3.5 if we do a 3.5.3 release.

mbalajew, 2024-07-02:
This is great, thank you. Did you need any help with testing?

dc42, 2024-07-04:
Fix now implemented in 3.5-dev source code too. @mbalajew which Duet are you using?

mbalajew, 2024-07-04:
@dc42 I'm using two Duet 3 Mini 5+ boards

dc42, 2024-09-23:
@mbalajew have you tested this fix in 3.5.3?

mbalajew, 2024-09-23:
Hey @dc42, yes, unfortunately it's still an issue on 3.5.3

droftarts, 2024-11-08:
I have tested this in both 3.5.3 and 3.6.0-beta.1, and both appear to work correctly for me, with the axes on the mainboard-as-expansion-board controlling normal axes.
@mbalajew are you controlling normal axes, or extruders, on the Mini 5+ expansion board?

Test setup: two Mini5+, running 3.6.0-beta.1. First board is on my normal bedslinger, second board has 1 motor attached to driver 0, and only has M954 A60 in its config.g. Configured motor as U axis, with same parameters as X axis.

Test: Initially M906 I30 and M84 S30 were set. I changed M84 S10, and idle timeout dropped to 10 seconds, on motors connected to both boards following a move of that axis (X and U tested). Changed M906 I100, motors on both boards maintain 100% current. Other M906 I# and M84 S# values tested, with the correct behaviour observed. So appears to be working correctly in 3.6.0-beta.1.
Reverted firmware to 3.5.3 on both boards, and ran tests again. Same result, so it seems to working as expected.

mbalajew, 2024-12-04:
Hi @dc42 and @droftarts, just a FYI - I’m still experiencing this issue. @droftarts and I have had several back-and-forth discussions on the Duet3D forums, but unfortunately, no resolution yet. At this point, I'm wondering if we should consider this as a potential hardware problem. I can confirm that the issue persists. I've already uploaded my configuration files to the forums and am happy to make a video demonstrating the problem if you think it would be helpful.

T3P3, 2024-12-05:
can you point me to the forum thread please.

mbalajew, 2024-12-05:
Hey @T3P3, here is the [thread](https://forum.duet3d.com/topic/35999/motor-current-idle-factor-not-working-over-canbus/28?_=1733413222948). It seems that since the team hasn't been able to reproduce my error on my specific hardware setup, they're considering the issue resolved.

I really don't want to come across as overly demanding - I fully appreciate how challenging it is to allocate resources in an open-source project. That said, the unexpected drop in idle current, particularly in setups like mine with a belt-driven z-stage, poses a significant safety risk. My printer has already sustained several damages because of this issue, and I'd hate for others to experience the same. I do have brakes on my z-stage motors but these will not help in this kind of scenario. 

Again, I deeply respect the effort it takes to maintain an open-source project, so please don't take this as an entitled complaint - just a genuine concern I felt was worth sharing.

T3P3, 2024-12-05:
Thanks @mbalajew We closed it as we replicated the issue then fixed it in the replication. as you say it may be a hardware issue or something else specific to you config. If its turns out be still a firmware issue we will reopen this to track it to resolution

droftarts, 2025-01-13:
Following my testing, there is an issue with setting M906 I parameter for expansion boards and mainboards-as-expansion. 

Tested in 3.5.3, 3.5.4 and 3.6.0-beta.2+5, on Mini 5+ WiFi v0.5 mainboard, Mini 5+ Ethernet v1.02a as expansion, and 1LC v1.2a (motor currents not tested on 1LC). X and Y axis are on mainboard, Z axis (4 motors) are on the mainboard-as-expansion.

Generally, sending M906 I# commands appear to be setting the motor idle current percentage correctly for me, in 3.5.3. However, it seems like the correct motor current is not restored when a move is made after the motor idle current is changed. For example, if M906 I1 (ie 1%) is set in config.g, turn on machine, G92 all axes, move Z, hold time and idle current are correct (same for X). If I then send M906 I100, followed by M906 I1, then move the Z axis, the motor current does not increase for the move or for the hold time. Same result in 3.5.3, 3.5.4 and 3.6.0-beta.2+5.

A similar issue occurs with expansion boards. Tested in 3.6.0-beta.2+5, on Mini 5+ WiFi v0.5 mainboard, 3HC v1.0, and 1LC v1.2a (motor currents not tested on 1LC). Though there is a difference in how this manifests itself. If the M906 I# is sent while the motor is on idle hold (ie at active motor power) the M906 I# command is ignored by the expansion board. Any subsequent moves happen at the motor current of the last registered M906 I# command, ie not at the 'proper' axis motor current.

closed 2025-01-17 by dc42