Duet3D/RepRapFirmware #1023: M17 doesn't re-engage motor brake

dc42, 2024-07-02:
In RRF 3.5.2 sending M17 doesn't re-engage the motor brake. This appears to be a consequence of the added support for brake PWM.

Forum thread: https://forum.duet3d.com/topic/33907/3-5-0-rc-1-brakes-not-engaging

dc42, 2024-07-02:
Comment from Andy S:

I had a (very) quick look at the code. Is the following code to enable the brake correct https://github.com/Duet3D/RepRapFirmware/blob/3.5-dev/src/Platform/Platform.cpp#L2731 if SUPPORT_BRAKE_PWM is enabled? I suspect it would not work on the STM32 port as once a pin has been set to some form of (hardware) PWM mode I do not think that calling writeDigital will override the PWM output. Should this instead be calling writeAnalog (as is used when disabling the brake)?

timschneider, 2024-07-04:
I was able to test if WriteAnalog(0.0) will work with brake on MB6HC.
https://github.com/Duet3D/Duet3Expansion/commit/3dad69662913b74810533de49a415c19c5e82959

```
 src/Platform/Platform.cpp | 4 +++-
 1 file changed, 3 insertions(+), 1 deletion(-)

diff --git a/src/Platform/Platform.cpp b/src/Platform/Platform.cpp
index 874b75ac..f3c2f560 100644
--- a/src/Platform/Platform.cpp
+++ b/src/Platform/Platform.cpp
@@ -2732,8 +2732,10 @@ void Platform::EngageBrake(size_t driver) noexcept
 {
 #if SUPPORT_BRAKE_PWM
 	currentBrakePwm[driver] = 0.0;
-#endif
+	brakePorts[driver].WriteAnalog(0.0);
+#else
 	brakePorts[driver].WriteDigital(false);			// turn the brake solenoid off to engage the brake
+#endif
 }
 
 void Platform::DisengageBrake(size_t driver) noexcept
```

and this works. The brake is now engaging and disengaging as expected.

dc42, 2024-07-04:
Thanks, Tim. I have implemented a similar fix in the 3.6-dev branch of RRF and I will backport it to 3.5-dev ready for the 3.5.3 release.

dc42, 2024-07-04:
Fix now implemented in 3.5-dev too.

closed 2024-09-12 by dc42