Duet3D/RepRapFirmware #1026: [FeatureRequest]: M6

Peppazzz91, 2024-07-15:
### Is your feature request related to a problem? Please describe.

Sice RRF has CNC mode, I use duet to control my cnc (PrintNC).
The duet can easily carry out all the operations of a domestic machine, so i don't need to move to a dedicated board.
But, one fundamental feature is missing: tool change

### Describe the solution you propose.

it would be sufficient for the duet, instead of returning an error if you give an M6 command, to look for an M6 macro in the Macro folder, so each user can write their own macro for their own machine

### Describe alternatives you've considered

Ex. A workaround exists but it is tedius, for example...

### Provide any additional context or information.

Ex. Photos, mockups, etc.

droftarts, 2024-07-15:
@Peppazzz91 Creating your own Gcode is a feature that already exists in RRF. Create a file called 'M6.g' in the /sys folder, populate it with the Gcodes you require. See https://docs.duet3d.com/en/User_manual/Reference/Gcodes#custom-gcodes

Peppazzz91, 2024-07-15:
Lol, i have tried with Macros folder and system/gcode folder.
Thanks then, I should have looked into the documentation better

closed 2024-07-15 by Peppazzz91