Duet3D/RepRapFirmware #1033: [FeatureRequest]: Multiple laser heads on single machine

nt7, 2024-08-19:
### Is your feature request related to a problem? Please describe.

With the tool-changing capabilities of the Duet3D firmware, we're building a printer with multiple laser heads and extruders.

We already figured out that we can swap between fff mode and laser mode when switching between the tools, but I'm struggling to define more than 1 laser head.

At the moment, these are the relevant Gcodes:

`M452 C"out9" R255 F200 ` enables Laser mode, on out9, with max intensity being 255, and a PWM frequency of 200

`M571 P"out9" F200 S255 `  during printing, this controls the laser based on E values in G1

There is no way to tell the FW to which printhead this laser belongs to.

### Describe the solution you propose.

It would be great if I had multiple M452 commands in my definition file
and then tie them to a tool number, similar to the way I define my other tools.

whenever I perform a tool-change, Duet should know which laser head to turn on when following G1 commands.

### Describe alternatives you've considered

The current workaround is to define the pin during the tool-change, then "undefine" the pin again before making the next tool-change. 
This feels very clunky and not very Duet-like.

### Provide any additional context or information.

Ex. Photos, mockups, etc.

T3P3, 2026-08-24:
we should move the port assignment as the same time, as per #842

T3P3, 2026-08-25:
if we implement this we need to also look at https://github.com/Duet3D/RepRapFirmware/issues/80, as there will be multiple tools that the output could be set via

T3P3, 2026-09-16:
Forum thread: https://forum.duet3d.com/topic/36349