Duet3D/RepRapFirmware #1039: [Bug]: Motor brake reports error when configured on mainboard used as expansion board

droftarts, 2024-09-09:
See from this and subsequent posts in the forum: https://forum.duet3d.com/post/344291 from this thread https://forum.duet3d.com/topic/36394/motor-brake-not-working-on-expansion-board

Motor brake reports error when configured on mainboard used as expansion board. User using two Mini5+ boards - the first as the main board and the second as the expansion board with a CAN address of 40. 

When configuring a brake on the mainboard-as-expansion board, an error is reported. Brake works correctly, it's just the error message.

```
M569.7 P40.3 C"out6"
Error: M569.7:
```

@dc42 said here https://forum.duet3d.com/post/344399
> It sounds to me that the code that supports a main board as an expansion board is returning an error code but otherwise working correctly.

mbalajew, 2024-09-09:
Hey @droftarts, I'm the user on the duet3d forums who originally posted this bug. Feel free to reach out if you need any help with testing

AndyEveritt, 2024-09-12:
@mbalajew can you try this 3.6-alpha FW and let me know if it fixes the issue.
[Duet3Firmware_Mini5plus.zip](https://github.com/user-attachments/files/16976556/Duet3Firmware_Mini5plus.zip)

mbalajew, 2024-09-12:
@AndyEveritt yes, I'll test this weekend. Thank you!

mbalajew, 2024-09-16:
@AndyEveritt issue is fixed with the alpha firmware you linked :+1:

closed 2024-09-16 by dc42