Duet3D/RepRapFirmware #1047: Startup issues when PanelDue and DWC are both active

dc42, 2024-09-20:
See https://forum.duet3d.com/post/345177. I believe I have a similar issue on my toolchanger.

T3P3, 2024-12-04:
@chrishamm has reproduced it.

We are proposing a change to PanelDue firmware to not request a new object model without there being no serial input for a e.g 150ms

dc42, 2024-12-09:
PanelDueFirmware updated as above, draft 3.5.1 release produced. Also added 'p' flag in M409 requests to indicate to RRF that the originator of the request is PanelDue, so that newer versions of RRF can omit data that isn't of interest to PanelDue from the response.

Tests on a toolchanger revealed that at 57600 baud the serial output stream to PanelDue was continuous using release 3.5.0 until the screensaver kicked in. In this draft release, when the machine is idle so only the frequently-changing fields of the OM are retrieved, the response is 750ms long with 200ms gaps between responses. When used with RRF 3.6.0-beta.2+5 which recognises the new 'p' flag, the response is shorted to 300ms.

closed 2025-04-12 by dc42