Duet3D/RepRapFirmware #1055: I want to use CNC and placement machine functions. I have an STM32 motherboard, model: Manta M8P v2.0+CB1. How should I compile the running firmware or system that supports this motherboard?

mayjack0312, 2024-10-26:
### Is your feature request related to a problem? Please describe.

I want to use CNC and placement machine functions. I have an STM32 motherboard, model: Manta M8P v2.0+CB1. How should I compile the running firmware or system that supports this motherboard?

### Describe the solution you propose.

The main control of the motherboard is: `STM32H723ZET6`, how should I transplant it correctly?

T3P3, 2024-10-26:
Please discuss this on the forum, In the STM section:
https://forum.duet3d.com/category/36/stm

closed 2024-10-26 by T3P3

jaysuk, 2024-10-26:
This board isn't supported by the STM32 port so it won't work

mayjack0312, 2024-10-26:
> Please discuss this on the forum, In the STM section: https://forum.duet3d.com/category/36/stm

OK, thank you for your reply

mayjack0312, 2024-10-26:
> This board isn't supported by the STM32 port so it won't work

Okay, then I will try other firmwares, thank you for your reply