Duet3D/RepRapFirmware #1061: Support using the scanning Z probe as a Z endstop by detecting when the reading stops changing

dc42, 2024-11-18:
The Beacon probe supports using the probe as a touch probe by moving the head towards the bed until the reading stops changing, indicating that contact has been made, We cold do the same with the scanning Z probe on the SZP board and the tool 1RR. We would need to implement this as a new probe mode in the expansion board firmware and have it signal the main board when the reading stops changing (or the rate of change decreases suddenly) after probing has started.

T3P3, 2024-11-21:
It would be helpful to be able to set the sensitivity of the response to allow people to find tune to their bed surface and probing speed

dc42, 2025-01-29:
Implemented experimentally in 3.6-dev. May need further refinement.

jbjhjm, 2025-02-03:
Thank you @dc42 ! This is great news for me!  😄

dc42, 2025-04-12:
This is working well now although it can be tricky to set up.

closed 2025-04-12 by dc42