Duet3D/RepRapFirmware #1063: Support Modbus sensors

dc42, 2024-11-26:
We should support temperature, humidity etc. sensors that communicate via Modbus. Examples:
- https://www.amazon.co.uk/Temperature-Humidity-Precision-Transmitter-Compatible/dp/B0836J2J2C. I have one of these. It supports Modbus function 4 to read register 1, register 2 or registers 1 and 2. Temperature is returned in register 1, humidity in register 2.
- https://uk.robotshop.com/products/industrial-grade-modbus-rs485-temperature-humidity-sensor
- https://www.calex.co.uk/product/temperature-measurement/infrared-temperature-sensors/pyrominibus/

We will need to specify at least the following:
- The slave address of the sensor
- Which function code to use (normally 3 or 4)
- Which register to read and how many registers to read
- Which register provides the required value and what scaling factor and offset to use
- The minimum interval required between successive reads

dc42, 2024-11-30:
I already added the framework for supporting virtual ports in RRF (similar to the existing code in Duet3Expanson) so we can add virtual Modbus ports in future.

T3P3, 2026-08-24:
related in concept to #718 - > reading from a bus into a variable

T3P3, 2026-09-14:
Forum threads:
https://forum.duet3d.com/topic/38319
https://forum.duet3d.com/topic/37759
https://forum.duet3d.com/topic/36986
https://forum.duet3d.com/topic/35901