Duet3D/RepRapFirmware #1065: [FeatureRequest]: Add Host Action support to M118

mlewis-everley, 2024-12-04:
### Is your feature request related to a problem? Please describe.

I really like RepRap firmware, but as I run a small print farm, I am heavily reliant on OctoPrint (as I imagine are a of other people in my position).

The ability to manage multiple instances via the OctoPrint API (using software like home assist or OctoFarm) is really helpful for managing multiple printers remotely, but on a fairly regular bases we might need to perform the following from the printer:

- Start a queued print (probably the most common)
- Pause a print
- Stop a print

My understanding is that OctoPrint can work well with RepRap boards, but currently you cannot get RepRap to tell the host to perform actions (such as the ones mentioned above) from the printer.

Looking at the RepRap docs, M118 is implemented (https://docs.duet3d.com/User_manual/Reference/Gcodes#m118-send-message-to-specific-target), but it is missing the `A` parameter (which is what marlin uses: https://marlinfw.org/docs/gcode/M118.html).

Realistically, this would most likely be performed from a menu item via a screen. But thanks to the awesome custom screen UI, this hopefully shouldn't be too much hassle to implement.

### Describe the solution you propose.

Add the ability to run `M118 A1 action:[action to send]` (as in Marlin: https://marlinfw.org/docs/gcode/M118.html) from a printer which would then send this command to the connected serial device.

### Describe alternatives you've considered

Cannot think of any?

### Provide any additional context or information.

RepRap M118 command: 
https://docs.duet3d.com/User_manual/Reference/Gcodes#m118-send-message-to-specific-target

Marlin M118 command:
https://marlinfw.org/docs/gcode/M118.html

T3P3, 2024-12-05:
How is this action actually sent - presumably as a string?

Looking at the octoprint documentation:
https://docs.octoprint.org/en/master/features/action_commands.html
The command is of the format:

`// action:command`

Have you tried sending:

M118 P1 S"// action:command"

(assuming octoprint is connected on the USB port so the flag P1 is used substitute for whatever connection is being used so the message goes to the correct target.

also do you have a forum thread discussing this?

mlewis-everley, 2024-12-06:
@T3P3 Thanks for the reply, unfortunately no forum thread. It was a conversation in the RepRapFirmware Discord group, I was asking about OctoPrint compatability and several people chimed in with words along the lines of "Works fine, but RepRap Firmware does not support host actions". Which led me to looking up `M118`.

I haven't tried `M118 P1 S"// action:command"` I haven't got a free OctoPrint instance right now, I will see if I can get one up and running and try it out.

mlewis-everley, 2025-01-09:
@T3P3 Just managed to get octoprint connected to one of my printers. You are correct, `M118 P1 S"// action:command"` works fine, so I have added some macros to a `Host` directory to handle start, cancel, pause and resume which has given me my marlin host actions back!

Many thanks for the help.

closed 2025-01-09 by mlewis-everley