Duet3D/RepRapFirmware #1066: [FeatureRequest]: MetaGCode Bitwise Math

bdistin, 2024-12-05:
### Is your feature request related to a problem? Please describe.

Missing mathematic operations that would allow some relevant programming paradigms to be used in meta-gcode.

### Describe the solution you propose.

It would be great if one could define an Int and track state using bitwise math. According to the docs, you can do (bool,bool)->bool with & and |; but this would allow you to do (int, int)->int with & and | too, as well as other bitwise operations.

### Describe alternatives you've considered

State tracking in MetaGcode for related states can currently be done with a vector of bools; however, the length has to be defined in advance and in likely different places in a meta gcode-base. It can also be done by writing the operations in base 10, but that becomes harder to track/easier to make a mistake.

### Provide any additional context or information.

Understandably, this is a little niche, and not everyone will make use of it. But here are a couple of examples to help convey how it could be used.

## Example 1:
### Only show the message if the 3rd bit of the global is 0 then set the 3rd bit to 1 after shown.

globals.g
```gcode
global tutorialMessagesDisplayedThisSession = 0
```
g37.g
```gcode
if { global.tutorialMode && global.tutorialMessagesDisplayedThisSession & 1 << 3 == 0  }
  M291 P{"The tool is about to move over the <b>ToolSetter<\b>! You will then be presented with a jog menu to <b>Manually<\b> decide where to measure from."} S4 K{"Continue", "Cancel"} F0
  set global.tutorialMessagesDisplayedThisSession |= 1 << 3
```

## Example 2:
### Store the current pin states before disabling them so they can be restored on resume.

globals.g
```gcode
global pinState = 0
```
pause.g
```gcode
var collector = 0

while { iterations < #state.gpOut }
  collector |= state.gpOut[iterations].pwm << iterations
  M42 P{iterations} S0

set global.pinState = collector
```
resume.g
```gcode
while { iterations < #state.gpOut }
  if { global.pinstate & 1 << iterations == 1 }
    M42 P{ iterations } S1
```

benagricola, 2024-12-05:
There's been a number of times bitwise operators on integers would have been helpful for me while writing meta gcode (similar to the above examples), and I think these will be useful when used alongside the new Modbus commands - some VFDs and other devices use bitmasked fields to store information.

It is possible to work with bitwise fields right now but it involves chaining existing maths functions.

```gcode

; By default, reset everything
var bitmask = 2

; Extract bits
if { mod(floor(var.bitmask/pow(2,0)),2) == 1 }
    echo { "Bit 0 set..."}
elif { mod(floor(var.bitmask/pow(2,1)),2) == 1 }
    echo { "Bit 1 set..."}
...
```

bdistin, 2024-12-13:
> Edit: I think it would also be fine to add these as functions rather than operators, which may be simpler from an implementation perspective.

Yeah, I would also be fine with functions. Examples were written using a preexisting/well-defined way so that the intention was (hopefully) more widely understandable.

Like I said in the OP:
> While specific operators or functions are up to your interpretation as it fits within the vision of RRF...