Duet3D/RepRapFirmware #1079: [FeatureRequest]: Use feedrate in heater model

AndyEveritt, 2025-02-06:
### Is your feature request related to a problem? Please describe.

Currently if you are extruding fast and trying to reach a higher target temperature (ie with `M309` temperature feedforward), the heating rate is lower than the model expects (because the filament is taking more of the heaters energy) which causes a heater fault to occur. I think the current extrusion rate needs to be accounted for when calculating the expected heating rate.

The value would probably need to be measured during tuning to be effective. Possibly at multiple speeds.

Currently I get heater faults multiple times per print when using heater feedforward (~10C) and printing fast.

The `M309 S` parameter only applies once the target temperature has been reached so is not currently relevant (or useful when paired with temperature feedforward?)

```cpp
void LocalHeater::SetExtrusionFeedForward(float pwmBoost, float tempBoost) noexcept
{
	if (mode == HeaterMode::stable)
	{
		const float pwmChange = pwmBoost - lastExtrusionPwmBoost;
		lastExtrusionPwmBoost = pwmBoost;
		InterruptCriticalSectionLocker lock;
		iAccumulator += pwmChange;
	}
	extrusionTemperatureBoost = tempBoost;
}
```

### Describe the solution you propose.

Possibly making the M309 pwmBoost apply during heating and stable sections would work

### Provide any additional context or information.

T3P3, 2026-09-14:
Forum threads:
https://forum.duet3d.com/topic/31284
https://forum.duet3d.com/topic/33600

T3P3, 2026-09-28:
@AndyEveritt do you still get heater faults with 3.7.0rc2 ? there have been fixes in this area.