Duet3D/RepRapFirmware #1082: SAME70 - Firmware update failures

T3P3, 2025-02-12:
Add an an additional erase in the IAP if all the write retries fail.

dc42, 2025-02-28:
Detail: occasionally the IAP fails to update the firmware on SAME70 processors. Re-flashing with Bossa succeeds. The limited debug information we have suggests that when this happens, the Write phase of the IAP failed on the same block after the maximum allowed retried. Our hypothesis is that the previous Erase step appeared to have worked, but the erase was incomplete. So our plan is that after a write command fails (after retries), we should repeat the Erase process, either for just the current segment or for the whole chip. If the error is that bits that should be 1 read back as 0 then there isn't even any need to retry the write process, becasuse writing can't change a 0 bit to 1.

dc42, 2025-04-25:
Updated the IAP so that when it is reading the firmware from SD card, if it fails to verify flash data that has been written because some bits read back as zero when they should be 1, it erases the current sector and starts writing from the start of the sector again, up to a maximum of 3 times. Doing this when the data source is DSF via SPI doesn't appear to me to be straightforward, so it is not implemented yet. Let's see whether this fixes the issue when updating firmware via the SD card.

dc42, 2025-04-30:
Christian updated the IAP to not check for write errors. If there is a CRC error at the end then the IAP retries the complete operation three times.

chrishamm, 2025-05-08:
Confirmed fixed after 200 firmware updates in SBC and standalone mode.

closed 2025-05-08 by chrishamm