Duet3D/RepRapFirmware #1085: Add support for CAN-FD bitrate switching

T3P3, 2025-02-12:
Optional for the system (i.e. done in config.g).

Related to
https://github.com/Duet3D/RepRapFirmware/issues/991

dc42, 2025-02-13:
We will need a facility for the main board to notify expansion boards whether BRS is to be used, and if so what the fast/normal data rate ratio is. One option for this is to always send time sync messages without using BRS and to use some of the 15 spare bits in the time sync message to specify whether BRS is in use and what the multiplier is. e.g. 0 = no BRS, 1 = x2, 2 = x3, 3 = x4. Alternatively, if we use a fixed ratio e.g. x4 then we can enable BRS on time sync messages too.

dc42, 2025-04-14:
We will not be able to use BRS when the main board is a Duet3 Mini (or anything else based on the SAME5x) because the datasheet says that the timestamp counter is not usable when BRS is used.

T3P3, 2026-10-07:
Evaluate and mitigate the impact on time sync of BRS on expansion boards.

dc42, 2026-11-06:
So far I have not been able to create increased jitter on a system using SAME5x expansion board with BRS enabled. I have changed the code so that the delay between receiving the time sync message and reading the step timer now depends only in the CAN interrupt latency and not also on the time taken to switch to the CanClock task. So unless there are situations in which the interrupt latency is unusually large, we should be able to use BRS with SAME5x and SAMC21 based expansion boards.

If in future we do detect situations in which BRS results in increased jitter then we can implement a mechanism to ignore a time sync message if the delay between the time stamp and reading the step clock is unusually large. This delay is mostly due to the time taken to receive the time sync message, since the time stamp related to the start of message reception but the step clock cant be read until the message has been fully received and the CAN peripheral generated an interrupt. Therefore, any new code to detect abnormally long delays must either be sensitive to the main CAN bit rate in use, or else must establish what a "normal" delay is and ignore time sync messages that have significantly longer delay.

dc42, 2026-11-16:
Still need to check intensive data collection from the expansion board.