Duet3D/RepRapFirmware #1087: [FeatureRequest]: Would like SD card access on 12864 in SBC configuration

jltx1, 2025-02-13:
### Is your feature request related to a problem? Please describe.

See: https://forum.duet3d.com/topic/32997/duet-3-sbc-with-12864-display-sd-card-access

I would also like access to the SD file list through the menu system on the 12864 display so I can initiate a print while physically standing in front of the printer.  Today I have to go to a different room where my computer sits.   

### Describe the solution you propose.

Ex. It would be great if...

### Describe alternatives you've considered

Ex. A workaround exists but it is tedius, for example...

### Provide any additional context or information.

Ex. Photos, mockups, etc.

dc42, 2025-03-03:
We are very unlikely to implement this request for the following reasons:

- It is difficult to implement because in SBC mode, GCode job files are read and pre-processed on the SBC. The existing code does not allow for running a job from local storage on the Duet.
- Many of the benefits of SBC mode (e.g. being able to intercept the GCode stream) would be lost when running a job from local storage
- It is not common to use a 12864 display in SBC mode. Normally the local interface would be a HDMI-connected touch screen attached to the SBC.

Have you considered adding a USB A socket connected to the SBC next to the 12864 display? You would then be able to print from a USB stick plugged in to that socket.

dc42, 2025-03-03:
Not planned

jltx1, 2025-03-03:
I think there is a misunderstanding of my request.  I specifically want a list of gcode files on the **SBC** SD card.  I do *not* want to launch a job from the Duet storage, nor the 12864 slot.  The SBC is where they are stored when sent over network and should be read from there.  But as it stands I cannot even get a listing of them on the 12864 display.  From the menu I attempt to execute an M32 as follows:

files R15 N4 I"0:gcodes" A"M32 #0|return|return"

the problem is that I cannot get a list of the files on the SD card mounted on the SBC.  By using M32 for a file on the SBC I should lose no benefits, right?   I disagree with your third point.  I use the 12864 all the time and I've always used an SBC.  There are many straightforward scenarios where this is very useful.

dc42, 2025-03-03:
OK, I did mis-understand. Re-opening this request. However, I still think the combination of a 12864 display and SBC mode is very unusual.