Duet3D/RepRapFirmware #1090: [Bug]: M558 not honoring M203

jltx1, 2025-02-16:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/37464/g30-travel-acceleration-and-speed?_=1739629006887

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [x] Duet3-Mini5+
- [x] Duet3-Mini2+
- [x] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.5.4

### Duet Web Control Version

3.5.4

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [x] Yes I use a SBC.
- [ ] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
M122
=== Diagnostics ===
RepRapFirmware for Duet 3 Mini 5+ version 3.5.4 (2024-11-24 10:44:24) running on Duet 3 Mini5plus WiFi (SBC mode)
Board ID: RNXYU-8296U-D65J0-40KMW-2503Z-ZP7HW
Used output buffers: 1 of 40 (17 max)
=== RTOS ===
Static ram: 103496
Dynamic ram: 116924 of which 0 recycled
Never used RAM 18212, free system stack 140 words
Tasks: SBC(2,rWait:,1.7%,813) HEAT(3,nWait 1,0.1%,323) Move(4,nWait 6,0.0%,256) CanReceiv(6,nWait 1,0.0%,939) CanSender(5,nWait 7,0.0%,336) CanClock(7,delaying,0.0%,334) TMC(4,nWait 6,1.5%,67) MAIN(2,running,95.5%,762) IDLE(0,ready,0.4%,29) AIN(4,delaying,0.8%,264), total 100.0%
Owned mutexes: HTTP(MAIN)
=== Platform ===
Last reset 00:32:59 ago, cause: power up
Last software reset at 2025-02-14 17:33, reason: User, Gcodes spinning, available RAM 17380, slot 1
Software reset code 0x6003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x00000000 BFAR 0xe000ed38 SP 0x00000000 Task SBC Freestk 0 n/a
Error status: 0x00
MCU revision 3, ADC conversions started 1484578, completed 1484576, timed out 0, errs 0
MCU temperature: min 16.1, current 29.2, max 34.4
Supply voltage: min 0.2, current 24.0, max 24.0, under voltage events: 0, over voltage events: 0, power good: yes
Heap OK, handles allocated/used 99/0, heap memory allocated/used/recyclable 2048/228/228, gc cycles 0
Events: 0 queued, 0 completed
Driver 0: standstill, SG min 0, read errors 0, write errors 0, ifcnt 12, reads 48739, writes 12, timeouts 0, DMA errors 0, CC errors 0
Driver 1: standstill, SG min 0, read errors 0, write errors 0, ifcnt 32, reads 48719, writes 32, timeouts 0, DMA errors 0, CC errors 0
[... 262 lines cut ...]
;use mechanical instead? G30 S-2 ; Z=0

; Restore high currents, speed & accel
M98 P"/macros/print_scripts/setup_printing.g"
```

### Details specific to your printer.

corexy, 3 z motors

### Links to additional info.

M558:
Tnnn Travel speed to and between probe points (mm/min). This is also the Z lift speed after probing. The corresponding axis speed limits set by M203 will be used instead if they are lower.

### What happened?

Expected result
valid travel acceleration and speed for my machine as set by M203,M201,M204

Observed result
M558 travel too fast, motors skipping

Steps to reproduce
Issue M558 without a T parameter.   Or set T parameter above M203 limit.

dc42, 2025-02-22:
M204 is intended to be used by slicers to control acceleration when running print jobs, so it is not a bug if probing sequences do not respect M204 limits. M203 limits should be respected during probing sequences. M201 limits should be respected during probing sequences, except that the actual probing move should respect the M201.1 limits (which should generally be lower than the M201 limits) instead.

jltx1, 2025-02-23:
Possibly related:  in 3.6.0-beta4 this behavior remains.  But also the very first move of the first M558 in bed.g above has an extreme jerk from standstill but subsequent moves appear as expected.  Maybe jerk limits ignored too?  I think the clue is that when I accidentally left off the T parameter it supplied its own travel speed (very high) from somewhere.  Maybe same with jerk and acceleration.

I have looked at mechanicals and everything is normal.  I have not observed this with other sudden moves from a stop.