Duet3D/RepRapFirmware #1094: [Bug]: Long homing moves can cause HeatTaskStuck software reset

gloomyandy, 2025-02-25:
A long homing move that results in the total number of steps exceeding 0x7ffffff can cause the processor to reset with a HeatTaskStuck reason. See this thread: https://forum.duet3d.com/topic/37165/issue-with-szp-on-secondary-idex-toolhead and also tests run by Jay when testing a mainboard in expansion mode.

This report was created at the request of DC42. His comments:

"moves that exceed 2^31 steps will cause problems, and probably always have although the symptoms may have changed. I guess we should add a check on the move length in G0 and G1 commands."

All of the examples I have seen have been as the result of a long homing move (often on Z as this tends to have a high step/mm). I suspect that in most cases other moves will have the length limited by axis length, but perhaps this is not being applied to "special" moves. It may also be possible to hit this problem with very long extrusion moves, not sure if we currently apply any sort of limit to them.

dc42, 2025-04-22:
Long homing moves can also cause Code 7 halts. See https://forum.duet3d.com/post/354537.

dc42, 2025-04-22:
We need to add two checks:
1. In each kinematics class, in CartesianToMotorSteps check that the final motor step position is in range and if not, return false. Ideally we would pass an error code back instead of just true/false.
2. In DDA:InitStandardMove, at the end check whether the move will take longer than 2^31 clocks. This could be done by having RecalculateMove return the duration in steps clocks as a float (in addition to storing it converted to int32_t as it already does) and checking that return value at the end of InitStandardMove.

dc42, 2025-04-23:
Fixed and tested.

closed 2025-04-23 by dc42