Duet3D/RepRapFirmware #1102: Extend filaments support to better support MMUs

dc42, 2025-03-31:
Currently, filaments are associated with extruders even though M700/701 act on the current tool. This doesn't work well with MMUs that can feed different filaments through a common extruder. The current work around is to map multiple extruder numbers t the same driver; however this causes issue when stall detection is used to load filament. See https://forum.duet3d.com/post/353387.

One way to improve on this would be:

1. Introduce the concept for feeders, where the combination of (extruder#, feeder#) defines a filament source and path.
2. M563 would take an optional additional parameter (default 0) specifying the feeder number for the first extruder number mentioned (or possibly the feeder numbers of all extruders mentioned).
3. The M701/702/703 filament management mechanism would map each (extruder, feeder#) combination to a filament instead of mapping just an extruder number to a filament. The filaments.csv file would be changed to reflect this.

T3P3, 2026-08-24:
related to #1234

T3P3, 2026-09-12:
Forum threads:
https://forum.duet3d.com/topic/37362
https://forum.duet3d.com/topic/36961
https://forum.duet3d.com/topic/33732
https://forum.duet3d.com/topic/32737