Duet3D/RepRapFirmware #1107: Add direct Modbus spindle control

dc42, 2025-04-18:
Many CNC spindle VFDs can be controlled over Modbus. Now that RRF supports Modbus we should extend M950 spindle configuration to support spindle speed and direction control over Modbus.

I think we will need to provide at least the following:
- A sequence of Modbus commands to configure the spindle controller and verify that it is responding to commands. This could be done in a macro file.
- A way of setting the speed and direction of the spindle. I don't expect that we will need to change the spindle speed/direction very rapidly, in which case this could be another macro.
- A way of stopping the spindle when emergency stop is used.

benagricola, 2025-04-18:
@dc42 great to see this on the horizon.

Based on my usage / implementation so far, there are some instances where data might need to be read from the VFD during configuration that can then be referenced when starting, stopping or changing RPM. A good example of this is with the Shihlin SL3 VFD that has a 'conversion factor' setting, which changes how the VFD accepts the input speed (a conversion factor of 1 means it accepts a frequency, 60 means it accepts an RPM).

Another example of this is the number of poles that the spindle has, which is also needed to calculate the correct frequency / RPM to send to the VFD.

I certainly can't see this being possible to do without using macros, and even if you ignore the non-modbus-compliant VFDs (like the Huanyang HY02D223B, which is basically ubiquitous on DIY CNCs), the specific Modbus / register implementation for each VFD is going to be different - managing C code to implement this will be prohibitive for supporting all of the VFDs that are in use I think, so yes - macros are the way to go.

Based on my work on https://github.com/MillenniumMachines/ArborCTL I would love to see a framework in RRF that formalises the "config" / "read data from VFD on startup" / "control" aspects of what I have implemented in ArborCTL.

How I achieve this at the moment is by having a 'wizard' process to pull in relevant VFD configuration parameters (spindle nameplate data mostly), which then executes a bunch of writes against the VFD to set the relevant settings.

There is then a daemon script that loads the required parameters from the VFD if not already loaded, and then monitors the state of one of the existing spindles in RRF, sending the relevant commands when the state changes and monitoring for unintended changes in spindle speed, as well as waiting for a spindle action to complete (i.e. waiting for a spindle to start up and reach target speed).

A very simple, very beneficial change to start with would be to allow a spindle to be configured without any pins. At the moment, we have to define a spindle with 3 pins to allow Modbus control with spindle direction, even though we are not using any of those pins.

I would say that we could get away with running spindle control commands once every 250-500ms. Emergency stop would have to bypass this but again, this will need to be a macro because triggering an e-stop will be different for each VFD.

And it should be clear that for true e-stop purposes, there should still be a physical e-stop circuit set up - either a contactor cutting power directly to the VFD or a physical input to a safety-rated VFD that triggers the estop.

Happy to discuss further if this brings up any questions either on the forum, teamgloomy or on Millennium Machines discord.

T3P3, 2026-09-08:
some related forum threads:
https://forum.duet3d.com/topic/37883/modbus-cnc-spindle-control-3-6-rc2-shihlin-sl3
https://forum.duet3d.com/topic/37792/hy-vfd-controlled-by-duet-3-mini
https://forum.duet3d.com/topic/36350/modbus-spindle-control

T3P3, 2026-09-12:
Forum thread: https://forum.duet3d.com/topic/33931