Duet3D/RepRapFirmware #1108: [FeatureRequest]: Support M73 C parameter

jltx1, 2025-04-19:
### Is your feature request related to a problem? Please describe.

I would like to know approximately when my print is going to pause for a color change or other due to M226.

### Describe the solution you propose.

Current slicers like PrusaSlicer and Orca slicer already emit this information in M73 using C parameter.  I would like RRF to process that and send it over to DWC and 12864 menu.

### Describe alternatives you've considered

stand around for hours and twiddle my thumbs

### Provide any additional context or information.

Ex. Photos, mockups, etc.

dc42, 2025-04-29:
Implemented experimentally in 3.6-dev in time for 3.6.0-rc.3. The estimated time to next user interaction as indicated in the M73 C parameter and corrected for the time elapsed since M73 with C parameter was last processed is now reported in the object model as `job.timesLeft.toPause`. Value is `null` if no information is available or the scheduled time has passed already, otherwise the estimated time in seconds. It's unlikely that DWC will be updated to display this value in the 3.6.0 release, however it should be available on the 12864 display and in DWC using the BtnCmd plugin.

dc42, 2025-04-29:
@jtx1 if we provide a new firmware binary and DWC, are you in a position to test this support? If so, which Duet do you have, and do you use it in standalone or SBC mode?

dc42, 2025-04-30:
Tested and working in 3.6.0-rc.3.

closed 2025-04-30 by dc42