Duet3D/RepRapFirmware #1112: [FeatureRequest]: Allow parameters to be passed as part of a tool change

jaysuk, 2025-05-06:
### Is your feature request related to a problem? Please describe.

It would be cleaner to allow parameters to be passed as part of a tool change call.
E.g. T0 A1 or T0 B"ABS"

### Describe the solution you propose.

This would be cleaner than having to pass any variables to an existing global before the tool change call

### Describe alternatives you've considered

Pass variables separately before the tool change call

### Provide any additional context or information.

Ex. Photos, mockups, etc.

T3P3, 2025-05-07:
@jaysuk can you expand on the usecase a bit further. what is the difficulty in plutting a command before the T command with the same information? Are you planning on making the custom T command in a post processor or is this a feature in some slicers?

T3P3, 2026-09-16:
Forum threads:
https://forum.duet3d.com/topic/37009
https://forum.duet3d.com/topic/37026