Duet3D/RepRapFirmware #1137: 3.6.0 - Z motor issues

T3P3, 2025-08-06:
Potentially related Z motor issues reported in 3.6.0:

https://forum.duet3d.com/topic/38506/3-6-0-3-z-axis-drops-after-start-of-print
https://forum.duet3d.com/topic/38279/z-steppers-losing-steps-randomly
https://forum.duet3d.com/topic/38181/z-axis-issues-with-3-6-0/37

Could be related to the motor current or phase issues. There is test code in 3.6.0+4 firmware.

dc42, 2025-08-09:
Investigating https://forum.duet3d.com/topic/38317/motor-direction-issue-after-upgrading-to-3-6-0/61 revealed that the mslutstart register appears to have been corrupted. The register is read-only so it's not easy to tell what value has been written to it, although that might be possible.

Possible causes:
1. Hardware issue, e.g. SCLK frequency too high, so the wrong register has been written.
2. Firmware corrupted the register number in the packet to he transmitted after the packet was set up but before it was transmitted.
3. DMA issue, e.g. the packet isn't entirely in non-cached memory.

dc42, 2025-08-11:
Reducing SPI clock frequency back to 2MHz as in 3.5.4 does not solve the issue.

dc42, 2025-08-11:
Investigation of the microstep table strongly suggests that the MSLUTSTART register has changed from its usual vale of 0x00F70000 to zero.